adafruit/Adafruit_nRF52_Bootloader #290: Tool to merge application firmware with bootloader for factory SWD/JLINK-programming

levpopov, 2023-01-26:
**Is your feature request related to a problem? Please describe.**
It'd be great to be able to flash both the bootloader and initial application firmware in one go when connected over SWD. AFAICT the only way to properly write bootloader settings is by actually performing a firmware update, which is inconvenient since this requires reconnecting target over USB (vs just using already connected SWD/JTAG).

**Describe the solution you'd like**
A CLI tool for merging application hex with bootloader hex that will configure the settings page to point to the bundled firmware. 

**Describe alternatives you've considered**
A potentially simpler option would be to just have a build flag for the bootloader to assume that a valid firmware has already been flashed (vs assuming empty). Application firmware would still be written over SWD but can be in a separate hex.

**Additional context**
N/A

vivek-pai, 2023-07-11:
I wrote a script to generate the correct bootloader settings from app firmware. With `mergehex`, I'm able to merge the bootloader, settings, and app into a single hex and flash that. I'm using it in my circuitpython build and it seems to be working great.

DanielCao0, 2024-11-26:
> I wrote a script to generate the correct bootloader settings from app firmware. With `mergehex`, I'm able to merge the bootloader, settings, and app into a single hex and flash that. I'm using it in my circuitpython build and it seems to be working great.

Hi , Can you share the merge method? Thank you.

closed 2026-05-15 by hathach