adafruit/Adafruit_nRF52_Bootloader #299: Make all filed with invalid suffix "c3a84" on integer constant

D4rk4, 2023-04-20:
### Operating System

Linux

### INFO_UF2.TXT

N/A

### What happened ?

```
make BOARD=metro_nrf52840_express all
CC main.c
src/main.c: In function 'main':
<command-line>: error: invalid suffix "c3a84" on integer constant
src/main.c:176:34: note: in expansion of macro 'MK_BOOTLOADER_VERSION'
  176 |   BOOTLOADER_VERSION_REGISTER = (MK_BOOTLOADER_VERSION);
      |                                  ^~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
make: *** [Makefile:412: _build/build-metro_nrf52840_express/main.o] Error 1

```

### How to reproduce ?

make BOARD=metro_nrf52840_express all

Also the same with the custom board based on nrf52840.

gamblor21, 2023-04-25:
I had the same issue came up. Looking into the makefile I see:
`-DMK_BOOTLOADER_VERSION=(80c3a84 << 16) + ( << 8) +  ` (note the two spaces after the second plus).

This is failing on line 176, as it obviously is not an actual integer.

I had not built my own bootloader for the nRF52840 prior so not sure what the actual value is supposed to be.

When I replaced the line with:
`BOOTLOADER_VERSION_REGISTER = 0x1234; //(MK_BOOTLOADER_VERSION);`
it did compile correctly.

hathach, 2023-04-26:
what is your output of following shell command
$ git describe --dirty --always --tags

also what is the output of following target:
$ make BOARD=metro_nrf52840_express print-GIT_VERSION

hathach, 2023-04-26:
mine is 
```
$ git describe --dirty --always --tags
0.7.0-17-g256e174-dirty
```

try to run `get fetch --tags` so that it could get the release version to pass to the code.

closed 2023-06-21 by hathach

Matheus-Garbelini, 2023-09-22:
@hathach indeed git fetch --tag was the solution. Hopefully this is added to the repo readme