adafruit/Adafruit_nRF52_Bootloader #301: `board_teardown()` resets every pin including the LDO pin on some boards

hyx0329, 2023-05-23:
## What happened

Current board config for Makerdiary M60 Keyboard has some flaws and I'm fixing them. The on-board power circuit requires an enable signal. By writing customized `board_init_extra()`, I enabled the pin, but it's reset by `board_teardown()`, and thus the keyboard powered off. I want to keep the power on.

I believe the board `challenger_840_ble` suffers the same issue. I took it as an example for `board_init_extra()`.

This can also be a feature request that asks for a method to retain the state of some specific pins. I know I can easily change the code to meet my need, but I want to discuss a little bit.

As for the fix for M60, I think I'll open a pr after resolving this issue.

## How to reproduce

1. enable LDO in `board_init_extra()`
2. boot the board without USB connection/power
3. the board won't boot(actually powered off quickly)

related code:

https://github.com/adafruit/Adafruit_nRF52_Bootloader/blob/80c3a8401bde50cd4795f4510a86149e118ed259/src/main.c#L178-L201

https://github.com/adafruit/Adafruit_nRF52_Bootloader/blob/80c3a8401bde50cd4795f4510a86149e118ed259/src/boards/boards.c#L156-L161

https://github.com/adafruit/Adafruit_nRF52_Bootloader/blob/80c3a8401bde50cd4795f4510a86149e118ed259/src/boards/challenger_840_ble/pinconfig.c#L22-L27

hathach, 2023-05-24:
could you enable it in application instead ?

hathach, 2023-05-25:
I see, can you make an PR to add board_init2() and board_teardown2() as weak function, and implemen it in your board. Those 2 function will be invoked right after the normal function with 
```C
if (board_init2) board_init2();
if  (board_teardown2) board_teardown2();
```

PR #305 (2023-05-26): Fix M60
- Change BUTTON_2 to ESC
- Always enable the battery

needs #304 
fixes #301
closed 2023-05-26 by hathach