adafruit/Adafruit_nRF52_Bootloader #306: DFU OTA always fails

hyx0329, 2023-05-26:
I observed that the DFU OTA always fails with latest code on my M60 keyboard(NRF52840). There is a solution not incorporated.

> The resolution was to modify src/sdk_config.h and increase the value of HCI_RX_BUF_QUEUE_SIZE to 16 resolved all issues.

_Originally posted by chrisingis in https://github.com/adafruit/Adafruit_nRF52_Bootloader/issues/35#issuecomment-518597429_

The old behavior was DFU App stucks at 1%(or fails) and the bootloader reboots to USB mode.

Extra info:
- Phone: Pixel 4XL
- OS: Lineage 20-20230507
- Nordic DFU App version 2.3.0 build 2212700
- bootloader commit e34b183f1d77ada051893bd6ee1b08660c292b78

_I need my patches to enter DFU OTA mode with buttons. That doesn't affect other functions._

jeffc, 2024-04-07:
One workaround for this is to go into the DFU app settings and turn on "packets receipt notification," then set the number of packets to something small. The Adafruit app (which does seem to work pretty reliably for DFU) [hardcodes 4 as a hack](https://github.com/adafruit/Bluefruit_LE_Connect_Android_V2/blob/ecb1ecdc550350956228b78c498a1bf875f99b00/app/src/main/java/com/adafruit/bluefruit/le/connect/dfu/DfuUpdater.java#L39)

ejtagle, 2026-01-15:
Could you try this pull request https://github.com/adafruit/Adafruit_nRF52_Bootloader/pull/375 to see if this fixes your issues?