adafruit/Adafruit_nRF52_Bootloader #327: Build without SD/BLE for nrf52820

ReFil, 2024-04-18:
I'm designing a custom PCB using the nRF52820 microcontroller, whilst largely similar to the 52840 and 52833 that are already supported it has a significantly compressed flash/ram space compared to these microcontrollers, such that the regular bootloader build would occupy more than 10% of the available flash space

As far as i can tell softdevice is needed only for BLE dfu, if that, and my application doesn't need softdevice, is it possible to modify the makefile to build without those features to use less flash space?

What other requirements are needed to port this bootloader to the 52820? I can see the `MCU_SUB_VARIANT` config which sets up the linker files and a few other requirements, the nrfx submodule needs updating too to bring in the new startup file for the 52820? I'm happy to take this task on myself, just need to understand the requirements a bit for porting to a new nrf microcontroller

jpconstantineau, 2024-04-19:
I am not sure about the 820 but the softdevice is essentially what's needed for any BLE functionality.  Think of it as a "BIOS" on a computer...  It interfaces with the radio on the chip and handles all the protocol details.  Without the softdevice, one would have to code all the protocol details - something that most would prefer to avoid. 
I suspect there is a dedicated softdevice for the 820 and one would need to get it and include it in the build for a 820 board.  Whether it supports the right version of  the nordic nrfx/sdk or not is another question for someone else to answer.

ReFil, 2024-04-19:
Zephyr has their own ble implementation that doesn't use soft device. My application is zephyr based. I do not want to include soft device as I need the flash discs occupied by it

kbladewht, 2024-05-04:
I am also seeking to have this supoort 52820 seems the library need to upgrade