adafruit/Adafruit_nRF52_Bootloader #334: Latest Version (0.9.1) Doesn't Work on Itsy Bitsy nrf52840

FoamyGuy, 2024-06-14:
### Operating System

Linux

### INFO_UF2.TXT

This one was working successfully:
```
UF2 Bootloader 0.8.3 lib/nrfx (v2.0.0) lib/tinyusb (0.12.0-145-g9775e7691) lib/uf2 (remotes/origin/configupdate-9-gadbb8c7)
Model: Adafruit ItsyBitsy nRF52840 Express
Board-ID: nRF52840-ItsyBitsy-revA
Date: Jan 29 2024
SoftDevice: S140 6.1.1
```

UF2 Bootloader 0.9.1 version was resulting in non-working circuitpython device. 

### What happened ?

Installing circuitpython via UF2 in version 0.9.1 does not seem to work. It does not report any errors, but it doesn't result in a working version of Circuitpython on the device. 

I ended up downgrading to bootloader version `0.8.3` and that one was able to successfully load the circuitpython UF2.

### How to reproduce ?

- Install bootloader version `0.9.1`
- open BOOT drive on computer and paste circuitpython UF2
- Paste finishes, device disconnects but then never reconnects. Nothing else happens, no blinks on RGB LED or anything, device is unresponsive.

SnappierSoap318, 2024-06-16:
How did you downgrade the Bootloader when the device was unresponsive?

FoamyGuy, 2024-06-16:
@SnappierSoap318 the device was seemingly unresponsive when it was trying to run CircuitPython, but double pressing reset still put it back into the bootloader mode. Once in the bootloader mode I used the same CLI instructions for upgrading bootloader to downgrade it by substituting in an older version zip file instead of the newest one.

dhalbert, 2024-06-17:
I tried this in several ways, on Ubuntu 24.04:
Updated a Feather nRF52840 from 0.8.3 to 0.9.0 and 0.9.1, with the update UF2. This worked fine.

Updated an ItsyBitsy nRF52840 from 0.2.13 to 0.9.1, using the latest adafruit-nrfutil. This worked, but the status RGB LED is not working. However I can see the BOOT drive, and I uploaded CircuitPython to 9.1.0-beta.3. I then updated with the 0.8.3 update UF2, and that worked also, but the status LED is still not working.

hathach, 2024-07-18:
hi @FoamyGuy I just pull out my itsybitsy which come with orginal bootloader v0.2.3. Then update it with arduion IDE to 0.8.x and using uf2 to bump to 0.9.0 and 0.9.1, then dropping cpy 9.1.0 and it works just fine. CPY loaded and everything seems to work just fine. Maybe trying it again, with simplest code.py or run storage erase first ? maybe you can try to flash it with arduino IDE, it will also reflash the softdevice just in case there is some corruption !!!?

```
UF2 Bootloader 0.9.1 lib/nrfx (v2.0.0) lib/tinyusb (0.12.0-145-g9775e7691) lib/uf2 (remotes/origin/configupdate-9-gadbb8c7)
Model: Adafruit ItsyBitsy nRF52840 Express
Board-ID: nRF52840-ItsyBitsy-revA
Date: Jun  5 2024
SoftDevice: S140 6.1.1
```

```
Adafruit CircuitPython 9.1.0 on 2024-07-10; Adafruit ItsyBitsy nRF52840 Express with nRF52840
Board ID:itsybitsy_nrf52840_express
UID:1572C0F4E6A10870
```

@dhalbert yeah the RGB LED doesn't work, it is incorrectly set to neopixel https://github.com/adafruit/Adafruit_nRF52_Bootloader/blob/master/src/boards/itsybitsy_nrf52840_express/board.h#L37 while the board actually use DOTSTAR. I think it is neopixel while prototyping, and isn't fixed since. I will made an PR to add dotstart support soon enough.

hathach, 2024-07-19:
#337 should fix itsybitsy dotstar led

hathach, 2024-08-28:
@FoamyGuy let me know if this is still an issue for you.

FoamyGuy, 2024-08-28:
This does seem to be resolved. I was able to use https://github.com/adafruit/Adafruit_nRF52_Bootloader/releases/download/0.9.2/update-itsybitsy_nrf52840_express_bootloader-0.9.2_nosd.uf2  to update the bootloader from `0.8.3` to `0.9.2` 

Then with the `9.2.0` bootloader I am able to successfully load different versions of circuitpython with UF2 files. 

The dotstar status RGB is working now too. 

Thank you @hathach!

closed 2024-08-28 by FoamyGuy

hathach, 2024-08-29:
thank you for your confirmation :)