adafruit/Adafruit_nRF52_Bootloader #338: Is there a way to rollback to previous firmware if OTA DFU fails?

Krupskis, 2024-07-22:
**Is your feature request related to a problem? Please describe.**
The issue with current OTA DFU is if it fails it will discard the main application and go into bootloader mode.

**Describe the solution you'd like**
Ideally, the bootloader should do a partition and rollback to previous firmware if OTA DFU fails.

Nianmi, 2024-12-19:
Hey @Krupskis 
Do we have an update on this? Any idea how we can solve this? I'm currently stuck at the same problem.

fanoush, 2025-02-03:
> if it fails it will discard the main application

It simply uses the same space and overwrites it. So it will 'discard' it no matter if it fails or not.

> Ideally, the bootloader should do a partition and rollback to previous firmware if OTA DFU fails.

that would mean you could only use half of the flash memory available

Nordic actually has an implementation of dual bank bootloader in their SDK but with this disadvantage - only half of flash can be used for uploading new and keeping old.

this bootloader is based on original Nordic single bank bootloader SDK source

hathach, 2025-02-06:
no way to roll back since this is single bank bootloader.

closed 2025-02-06 by hathach