adafruit/Adafruit_nRF52_Bootloader #339: Compile error - array subscript

borisbme, 2024-07-22:
### Operating System

Linux

### INFO_UF2.TXT

I have no info_uf2.txt

### What happened ?

Installed all perquisites, tried to make the bootloader for one board, and it failed
Using Manjaro, latest kernel, latest updates

make BOARD=feather_nrf52840_express all

### How to reproduce ?

Issue the  make BOARD=feather_nrf52840_express all on a Manjaro system

### Debug Log

`make BOARD=feather_nrf52840_express all

CC dfu_ble_svc.c
CC dfu_init.c
CC flash_nrf5x.c
CC main.c
CC screen.c
CC images.c
CC boards.c
CC nrfx_power.c
CC nrfx_nvmc.c
CC system_nrf52840.c
CC bootloader.c
CC bootloader_settings.c
lib/sdk11/components/libraries/bootloader_dfu/bootloader_settings.c: In function 'bootloader_mbr_addrs_populate':
lib/sdk11/components/libraries/bootloader_dfu/bootloader_settings.c:45:7: error: array subscript 0 is outside array bounds of 'const uint32_t[0]' {aka 'const long unsigned int[]'} [-Werror=array-bounds=]
   45 |   if (*(const uint32_t *)MBR_BOOTLOADER_ADDR == 0xFFFFFFFF)
compilation terminated due to -Wfatal-errors.
cc1: all warnings being treated as errors
make: *** [Makefile:412: _build/build-feather_nrf52840_express/bootloader_settings.o] Error 1`

iffycodes, 2024-08-18:
I am seeing the same issue building with the nRF Connect SDK v2.5.0...

Makefile:392: update target '_build/build-wio_tracker_1110/bootloader_settings.o' due to: lib/sdk11/components/libraries/bootloader_dfu/bootloader_settings.c
echo CC bootloader_settings.c
CC bootloader_settings.c
arm-none-eabi-gcc -DNRF52840_XXAA -DS140 -DDFU_APP_DATA_RESERVED=10*4096 -mthumb -mabi=aapcs -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -ggdb -Os -ffunction-sections -fdata-sections -fno-builtin -fshort-enums -fstack-usage -fno-strict-aliasing -Wall -Wextra -Werror -Wfatal-errors -Werror-implicit-function-declaration -Wfloat-equal -Wundef -Wshadow -Wwrite-strings -Wsign-compare -Wmissing-format-attribute -Wno-endif-labels -Wunreachable-code -Wno-unused-parameter -Wno-expansion-to-defined -fno-ipa-modref -D__HEAP_SIZE=0 -DCONFIG_GPIO_AS_PINRESET -DCONFIG_NFCT_PINS_AS_GPIOS -DSOFTDEVICE_PRESENT -DUF2_VERSION='"0.7.0 lib/nrfx (v2.0.0) lib/tinyusb (0.12.0-145-g9775e7691) lib/uf2 (remotes/origin/configupdate-9-gadbb8c7)"' -DBLEDIS_FW_VERSION='"0.7.0 s140 6.1.1"' -DMK_BOOTLOADER_VERSION='(0 << 16) + (7 << 8) + 0' -Isrc -Isrc/boards -Isrc/boards/wio_tracker_1110 -Isrc/cmsis/include -Isrc/usb -Ilib/tinyusb/src -Ilib/nrfx -Ilib/nrfx/mdk -Ilib/nrfx/hal -Ilib/nrfx/drivers/include -Ilib/nrfx/drivers/src -Ilib/sdk11/components/libraries/bootloader_dfu/hci_transport -Ilib/sdk11/components/libraries/bootloader_dfu -Ilib/sdk11/components/drivers_nrf/pstorage -Ilib/sdk11/components/ble/common -Ilib/sdk11/components/ble/ble_services/ble_dfu -Ilib/sdk11/components/ble/ble_services/ble_dis -Ilib/sdk/components/libraries/timer -Ilib/sdk/components/libraries/scheduler -Ilib/sdk/components/libraries/crc16 -Ilib/sdk/components/libraries/util -Ilib/sdk/components/libraries/hci/config -Ilib/sdk/components/libraries/uart -Ilib/sdk/components/libraries/hci -Ilib/sdk/components/drivers_nrf/delay -Ilib/softdevice/s140_nrf52_6.1.1/s140_nrf52_6.1.1_API/include -Ilib/softdevice/s140_nrf52_6.1.1/s140_nrf52_6.1.1_API/include/nrf52 -c -o _build/build-wio_tracker_1110/bootloader_settings.o lib/sdk11/components/libraries/bootloader_dfu/bootloader_settings.c
lib/sdk11/components/libraries/bootloader_dfu/bootloader_settings.c: In function 'bootloader_mbr_addrs_populate':
lib/sdk11/components/libraries/bootloader_dfu/bootloader_settings.c:45:7: error: array subscript 0 is outside array bounds of 'const uint32_t[0]' {aka 'const long unsigned int[]'} [-Werror=array-bounds=]
   45 |   if (*(const uint32_t *)MBR_BOOTLOADER_ADDR == 0xFFFFFFFF)
compilation terminated due to -Wfatal-errors.
cc1: all warnings being treated as errors

hathach, 2024-09-05:
what is your `arm-none-eabi-gcc --version` output

iffycodes, 2024-09-07:
(redboard-west) b00p@lappy:~/workspaces/work-lora/redboard/ncs/sidewalk/samples/wm1110-asset-tracker/bootloader/Adafruit_nRF52_Bootloader$ arm-none-eabi-gcc --version 
arm-none-eabi-gcc (15:13.2.rel1-2) 13.2.1 20231009
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

iffycodes, 2024-09-12:
Hmmmm. When I checked the version, my environment was default. When I had the error, I might have been using a different build environment. These are the gcc versions I have on my system beyond 13.2.1....

root@lappy:/# ./home/b00p/SimplicityStudio5/SimplicityStudio_v5/developer/toolchains/gnu_arm/12.2.rel1_2023.7/bin/arm-none-eabi-gcc --version
arm-none-eabi-gcc (Arm GNU Toolchain 12.2.Rel1 (Build arm-12.24)) 12.2.1 20221205
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

root@lappy:/# ./home/b00p/SimplicityStudio5/SimplicityStudio_v5/developer/toolchains/gnu_arm/10.3_2021.10/bin/arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10.3-2021.10) 10.3.1 20210824 (release)
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not everoot@lappy:/# ./home/b00p/.particle/toolchains/gcc-arm/10.2.1/bin/arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10-2020-q4-major) 10.2.1 20201103 (release)
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
n for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Those are all the arm gcc versions I have beyond 13.2.1. 
At the end of the day, I was able to build and link by removing -Werror from the makefile and it seems to be ok.

hathach, 2024-09-13:
what is your git hash, maybe try to update if it isn't latest

borisbme, 2024-10-06:
> Hmmmm. When I checked the version, my environment was default. When I had the error, I might have been using a different build environment. These are the gcc versions I have on my system beyond [...]

Same here,
Using latest GCC

>>arm-none-eabi-gcc --version

arm-none-eabi-gcc (Arch Repository) 14.1.0
Copyright (C) 2024 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

compiles only if I delete -Werror from the makefile

jgartrel, 2024-12-05:
It looks like this was introduced as a feature in GCC v12: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=105523 .

jgartrel, 2024-12-05:
It looks like it can be resolved by adding `--param=min-pagesize=0` to the `Makefile`:
```
diff --git a/Makefile b/Makefile
index 50b8a3b..48fa27c 100644
--- a/Makefile
+++ b/Makefile
@@ -262,6 +262,7 @@ CFLAGS += \
        -Wmissing-format-attribute \
        -Wno-endif-labels \
        -Wunreachable-code \
+       --param=min-pagesize=0 \^M
        -ggdb
 
 # Suppress warning caused by SDK
```

JzPyro, 2024-12-16:
This error persists in version 14 It is fixed by adding 14.% to the Makefile on lines 340-342 for me

`ifneq (,$(filter 12.% 13.% 14.%,$(shell $(CC) -dumpversion 2>/dev/null)))
	CFLAGS += --param=min-pagesize=0
endif`

andreaskurz, 2025-05-05:
@hathach I made a PR for the fix but without these guys here I'd also wonder why it wouldn't compile – so thanks to the people who proposed solutions here :)

hathach, 2025-05-07:
thank @andreaskurz this should be fixed by #353

closed 2025-05-07 by hathach