adafruit/Adafruit_nRF52_Bootloader #345: nice nano v2 upload problem

goktugh, 2024-12-21:
i am using nice nano v2, burn bootloader works perfect but when I try to upload a sketch, I get this error

Failed to upgrade target. Error is: ClearCommError failed (PermissionError(13, 'The device does not recognize the command.', None, 22))
Traceback (most recent call last):
  File "__main__.py", line 296, in serial
  File "dfu\dfu.py", line 235, in dfu_send_images
  File "dfu\dfu.py", line 203, in _dfu_send_image
  File "dfu\dfu_transport_serial.py", line 155, in send_init_packet
  File "dfu\dfu_transport_serial.py", line 243, in send_packet
  File "dfu\dfu_transport_serial.py", line 264, in get_ack_nr
  File "serial\serialwin32.py", line 275, in read
serial.serialutil.SerialException: ClearCommError failed (PermissionError(13, 'The device does not recognize the command.', None, 22))

Possible causes:
- Selected Bootloader version does not match the one on Bluefruit device.
    Please upgrade the Bootloader or select correct version in Tools->Bootloader.
- Baud rate must be 115200, Flow control must be off.
- Target is not in DFU mode. Ground DFU pin and RESET and release both to enter DFU mode.

hathach, 2024-12-25:
which PC OS you are using, look like you have com permission issue. If it is macos/windows, try to run with root/admin right, for linux you need to update udev rules

goktugh, 2024-12-25:
> which PC OS you are using, look like you have com permission issue. If it is macos/windows, try to run with root/admin right, for linux you need to update udev rules

I am using Windows and have started Arduino with admin permissions. again no luck

MainTobias, 2025-01-20:
@goktugh have you had any success yet?

I facing a similar issue where I copy the file over but it doesn't flash successfully.

However I have yet to find a way to get debug messages.

MainTobias, 2025-03-08:
@goktugh have you fixed it?